ComputationalRadiationPhysics/jungfrau-photoncounter #50: Publications

heidemeissner, 2019-02-22:
Hi all,
we have some questions concerning the scientific outcome of our project:
(1) Is it ok for you that we show images of the Jungfrau detector on a poster and in a talk on the Helmholtz-Matter and Technology Meeting? 
(2) We plan to publish our results in Journal of Computational Physics including the benchmark tests. Are you also planning to publish s.th. on your methods / detector data that we can use as reference? Or would you like to add a section to our paper? That would be perfect, then we'd have a common publication.
Have a nice day,
Heide

sredford, 2019-02-28:
Hi Heide,

1) no problem to show some pictures. Do you have some? Otherwise I can send you a couple.

2) We (Bernd, Carlos, Martin, Sophie) agree with you that a common publication would be the best outcome. That way we can introduce and motivate the use of JF, describe the 'problems', explain the GPU solution and talk about how this can be integrated in a whole DAQ system. We already have a few references that can be included.
Do you have a suggestion how we can all work together on such a document? Github, overleaf, googledoc?

Cheers, Sophie

heidemeissner, 2019-03-08:
Dear Sophie,
(1) great, thanks. 
(2) Fantastic. Maybe Gitlab or overleaf would be best, I am in contact with the computational science department for further options. It should be a private repository. Do you prefer Word or Latex?
(2b) From HZDR we have Jonas, Florian, Michael and me. Would it be ok, if one of you is first author and Michael or myself last one? What would you suggest?
Cheers, Heide

sredford, 2019-03-21:
Dear Heide,
We prefer Latex.
From PSI we have Carlos, Sophie, Martin, Anna, Aldo, Bernd.
We don't mind about the order. What you suggest is fine.
Cheers, Sophie