ComputationalRadiationPhysics/jungfrau-photoncounter #52: Using RMS for pixel masking

kloppstock, 2019-03-13:
As discussed in the previous Skype meeting, we are in the process of implementing an RMS based check and mask pixels which are over a certain threshold. While doing this however, some questions came up:
* Should this RMS value only be calculated over the initial moving window or over the whole calibration phase? Should the standard deviation be handled in a similar way?
* Should this RMS value be calculated from raw ADC input or from something different?
* Would a high standard deviation already indicate that a pixel should be masked?

sredford, 2019-03-19:
Hi Jonas,

- the mask on the pedestal RMS should be calculated at the end of the dark frames.  After that, since there is a threshold on the RMS before the frame is used in the pedestal calculation, the reported RMS is no longer accurate (it should start becoming smaller).
- the RMS value can be calculated in raw ADC, that's easiest.
- I think we use the terms RMS and standard deviation interchangeably, to mean the sqrt of the variance. The definitions (as we use them) are in jungfrauPedestal.C and MovingStat.h. Do you think we have a mistake in the terminology?

Cheers, Sophie

TheFl0w, 2019-03-21:
I think the RMS is not necessarily the same as standard deviation unless the mean is 0. Apparently the standard deviation (sigma) is also called RMSE (root mean square error) or RMSD (root mean square deviation). Our implementation always used the square root of the variance for sigma instead of the RMS so we were wondering if you wanted us to change it. I guess we already used the correct one.
[image]