ComputationalRadiationPhysics/jungfrau-photoncounter #57: Compilation errors

j-stephan, 2019-06-26:
The latest commit broke a few things.

While compiling `main.cpp`:

```
jungfrau-photoncounter/include/jungfrau-photoncounter/Config.hpp:74:1: error: unknown type name 'i'
i    // the struct for the initial pedestal data
^
jungfrau-photoncounter/include/jungfrau-photoncounter/Config.hpp:75:5: error: expected member name or ';' after declaration specifiers
    struct InitPedestal {
    ^
jungfrau-photoncounter/include/jungfrau-photoncounter/Config.hpp:179:40: error: use of undeclared identifier 'InitPedestal'
    using InitPedestalMap = CheapArray<InitPedestal, DIMX * DIMY>;
                                       ^
In file included from main.cpp:3:
In file included from jungfrau-photoncounter/include/jungfrau-photoncounter/Dispenser.hpp:17:
jungfrau-photoncounter/include/jungfrau-photoncounter/kernel/Summation.hpp:2:10: fatal error: 'AlpakaHelper.hpp' file not found
#include "AlpakaHelper.hpp"
         ^~~~~~~~~~~~~~~~~~
4 errors generated.
```

kloppstock, 2019-06-26:
Thanks for mentioning it. It was just a typo in the Config.hpp which is fixed now. 
Could you try compiling it again?

j-stephan, 2019-06-27:
Yup, just facing issues in the SYCL backend now. Thanks!

commit 8b3f216a (2019-07-03): Implemented first test benchmark using google benchmark. Also fixes #57.
closed 2019-06-27 by j-stephan