ComputationalRadiationPhysics/jungfrau-photoncounter #62: Reference Data

kloppstock, 2019-10-10:
While debugging, I noticed that a significant amount of pixels in the last image of the MOENCH data set would have a negative photon count. To better understand this problem it would be good to know, what should be visible in the MOENCH data set and where the beam hit the detector. If you have any reference images from this data set, we would be happy to compare our results to them. 

I have the results of the last images (of the MOENCH as well as of the Jungfrau data set) attached. Do they look otherwise reasonable?

Cheers, Jonas

**MOENCH Data**

[1] Map of Negative Photon Values of Frame 30 000 (Legend: 1 - positive, 0 - 0, -1 - negative)
[image]

[2] Energy Data of Frame 30 000 (Scaled Logarithmically)
[image]

[3] Photon Data of Frame 30 000 (Scaled Logarithmically)
[image]

[4] Photon Data of Frame 30 000
[image]

[5] Photon Data of Frame 30 000 (Negative Values Clipped to 0)
[image]

[6] Pedestal Values after Frame 30 000
[image]

[7] Drift of Pedestal Values until Frame 30 000
[image]

[8] Drift of Pedestal Values until Frame 30 000 (Scaled Logarithmically)
[image]

**Jungfrau Data**

[1] Map of Negative Photon Values of Frame 10 000 (Legend: 1 - positive, 0 - 0, -1 - negative)
[image]

[2] Energy Data of Frame 10 000 (Scaled Logarithmically)
[image]

[3] Photon Data of Frame 10 000 (Scaled Logarithmically)
[image]

[4] Photon Data of Frame 10 000
[image]

[5] Photon Data of Frame 10 000 (Negative Values Clipped to 0)
[image]

[6] Pedestal Values after Frame 10 000
[image]

[7] Drift of Pedestal Values until Frame 10 000
[image]

[8] Drift of Pedestal Values until Frame 10 000 (Scaled Logarithmically)
[image]

lopez-c, 2019-10-16:
Hello Jonas,

Yes, as you mentioned in your message it is strange to have negative photon count values, in fact they should not be there unless the pixel giving those negative photon counts is defective. 

First of all, from what we can see (the scale range and the colours do not let us distinguish properly photon data and pedestal drifts in linear scale), the images for the Jungfrau dataset looks reasonable. That said, let's try to understand what can go wrong with the Moench dataset.

For the Moench dataset, the beam hits the detector in the center of the detector, you can cleraly see it in the images where you plot the drift of the pedestal, as well as in the photon data plots (with a bit more difficulties). The effect that you see at the top of the detector (rows 320 - 400) is a defect in the detector and not the beam.

The amount of negative photon count values, as well as the fact that some pixels count much more than 1 photon, makes us think that there could be something wrong with the coefficients used to compute the photon counts. 

Which gain coeffcients are you using? An approximate value that should be enough for this measurements (until we have calibrated values for each Moench detector) is a gain factor of 150ADU/keV ( as a comparison, the values used for the first gain stage of Jungfrau were around 40 ADU/keV). 

Another point is to know which beam energy are you using. The dataset for Moench was recorded with a beam energy of 8.7 keV i.e. the total cluster should sum up to about 1305 ADU. 

If these two values, gain and beam energy match, the problem could come from a wrong update of the pedestal. In fact, unlike in Jungfrau where (almost) each photon hits only one pixel, in Moench the hit is shared among several neighboring pixels, and the pedestal should not be updated for those neighboring pixels, which I don't know if it is the case. 

After all, the RMS of pixels without photons should be about 25 ADUs, so some negative values can be present after pedestal subtraction, but they should rarely be lower than -100 ADUs and never correspond to an entire photon, unless the pixel is bad as we said. 

Let's see if we can clarify this 3-4 points.

Thanks again for all your effort.

Best regards,
 Anna and Carlos

kloppstock, 2019-10-16:
Hello Anna, hello Carlos,

since the beam energy wasn't mentioned, we used the same as for the Jungfrau data set (which should be 12.4keV). For the gain we assumed a constant gain of 1. Is the gain the same for each pixel or is there a gain map required similar to the Jungfrau data set?

Anyway, I will adapt the code and upload new images once they are ready. 

Best regards, 
Jonas

lopez-c, 2019-10-16:
Hello Jonas,
You can take a constant gain factor of 150ADU/keV for all the pixels (or create a gain map with that value). As I was saying, there is not a detailed calibration for that detector and these 150ADU/keV is a good approximation.
Best regards,
 Carlos

kloppstock, 2019-10-21:
Hello Carlos, 

I have adapted the parameters for the MOENCH data set and the results do look a lot more reasonable now. 

Cheers, 
Jonas

[1] Map of Negative Photon Values of Frame 30 000 (Legend: 1 - positive, 0 - 0, -1 - negative)
[image]

[2] Energy Data of Frame 30 000 (Scaled Logarithmically)
[image]

[3] Photon Data of Frame 30 000 (Scaled Logarithmically)
[image]

[4] Pedestal Values after Frame 30 000
[image]

[5] Drift of Pedestal Values until Frame 30 000 (Scaled Logarithmically)
[image]

closed 2020-03-02 by kloppstock