ComputationalRadiationPhysics/jungfrau-photoncounter #66: Clustering different on backends and nodes

heidemeissner, 2020-02-03:
Hi Jonas,

I plotted the data in data.zip and they pretty much look the same. Just the average is very slightly different. What is wrong?

Cheers,
Heide

kloppstock, 2020-02-03:
My guess is that the pedestal update is not correctly working which results in a small drift in the pedestal and energy data. Unfortunately, I don't know yet what exactly is causing this behavior. 

Cheers, 
Jonas

heidemeissner, 2020-02-03:
So the problem is not due to clustering? Are the data you sent clustered? What are the numbers of clusters for them? Do they differ much? Inside data.zip the difference between results and reference are that tiny, that it can be ignored.

heidemeissner, 2020-02-04:
Please test the following:
- switch clustering off and test if results, i.e. energy maps, are ok without clustering
- start with serial computation on CPU, then parallel CPU, then CUDA on fwk?? and hemera
- provide text files of energy map at all update events

kloppstock, 2020-02-04:
Here is an update on the current situation:

I have tested all relevant configurations (including clustering) on the FWK394 node (both on CPU and GPU). Everything except clustering on the GPU seems to be working. The clustering results are slightly off (energy values off by at most 2 and cluster count off by at most to every 10 frames most of the time). These results change slightly every run so I assume that there is a race condition or a scheduling problem. If only one GPU with one stream is present, it also works as expected. Turning off pedestal updates also yields in the correct results which hints on problems with the copying of the pedestal data between accelerators or the pedestal update implementation itself. 

Cheers,
Jonas

heidemeissner, 2020-02-05:
Hi Jonas,

ok, thanks. That means that the bug is localized. It can be due to miss-initialized memory or copying processes which go wrong. Please check, if you have really one stream. Furthermore, you can set CUDA_LAUNCH_Blocking=1 and test if the bug still occurs.

Cheers,
Heide

kloppstock, 2020-02-17:
Hello Heide, 

today I ran the program with the blocking flag set but unfortunately this didn't help. I also used the cuda-memcheck tool to look for illegal memory accesses, race conditions, synchronization problems and uninitialized data. I found out, that the pedestal maps for gain stage 1 and 3 were not correctly initialized. After fixing this, the problem unfortunately still persists even though non of the above mentioned tools complain anymore. 

Cheers, 
Jonas

heidemeissner, 2020-02-19:
Hi Jonas,
could you tell me in which configuration the problem occurs? CPU, GPU, CUDA,...clustering on or off and include the results (images where the bug, i.e. different values are obvious)?
And please also attach the links to the branches you used.
I'm not at HZDR next week, except of Friday morning, but available by mail. 
Cheers,
Heide

kloppstock, 2020-02-24:
Hi Heide,
the problem only occurs when using Clustering on CUDA with multiple devices, masking enabled and pedestal updates implemented. The reference implementation is located in the [reference-tmp](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/tree/reference-tmp) and the (wrong) results were generated by the code in the [clustering-bug](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/tree/clustering-bug) branch. The output data is located in the [data.zip file in the clustering-bug branch](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/blob/clustering-bug/data.zip). The ```pedestal_n.txt``` files are read out after each step. The ```pedestal-first.txt``` file contains the initial pedestal maps. This data suggests that the problem might be caused by incorrect pedestal updates since the first energy map is still correct but the first step already results in differently updated pedestal maps. 

My next try would be using the clustering kernel from the reference implementation (without the element layer) and see if it works with this. 
Cheers, 
Jonas

Command line output from the reference:
[reference.txt](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/files/4244638/reference.txt)

Command line output from incorrectly working implementation:
[result.txt](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/files/4244639/result.txt)

kloppstock, 2020-02-24:
Hello Heide, 
I tested the other kernel (without the element layer) and this seemed to work for the problematic configuration. Therefore I have integrated the new and the old kernel and the old kernel and select between them depending on the used device which seems to work on FWK394 as well as on hemera. These changes are now also on the [clustering-bug](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/tree/clustering-bug) branch in the [ClusterFinder.hpp](https://github.com/ComputationalRadiationPhysics/jungfrau-photoncounter/blob/clustering-bug/include/jungfrau-photoncounter/kernel/ClusterFinder.hpp) file.
Cheers, 
Jonas

closed 2020-03-02 by kloppstock